dqrobotics/python #1: Import error in Python 3

juanjqo, 2019-04-03:
Hi @mmmarinho, I was using the DQ library for Python 2. Now, I'm migrating to Python 3, since python2 will not be supported anymore. 
I installed the library ( python3 -m pip install dqrobotics ). But I have an import error.

In a new terminal:

> juanjqo@pop-os:~$ python3

> Python 3.6.7 (default, Oct 22 2018, 11:32:17) 
> [GCC 8.2.0] on linux
> Type "help", "copyright", "credits" or "license" for more information.

>  from dqrobotics import*

> Traceback (most recent call last):
> File "<stdin>", line 1, in <module>
> ImportError: /home/juanjqo/.local/lib/python3.6/site-packages/dqrobotics.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN11DQ_robotics31line_to_point_distance_jacobianERKN5Eigen6MatrixIdLin1ELin1ELi0ELin1ELin1EEERKNS_2DQES7_

Can you help me?

Best regards, 

Juancho

mmmarinho, 2019-04-04:
Hello, @juanjqo!

There is nothing wrong at your end, it was a mistake I made in the C++ implementation.

I pushed a fix and I can import the package without issues in my 18.04. Just update the package and you should be ready to go
```
python3 -m pip install dqrobotics --upgrade
```

closed 2019-04-04 by mmmarinho

juanjqo, 2019-04-04:
Thanks @mmmarinho 
It is working!