dqrobotics/python #4: Missing attributes in DQ_SerialManipulator

juanjqo, 2019-04-22:
Hi @mmmarinho 

I had working a kinematic control in Python two weeks ago. The last version breaks my current implementation. I understand that DQ robotics is experimental in Python and I'm updating my code. However, some attributes are missing in DQ_SerialManipulator, specifically:

- reference_frame 

- base_frame 

- n_links  

Of course, the legacy attribute 'set_base' is not working either. 

Version: dqrobotics 0.1.1.10
Ubuntu 18.04 LTS

Best regards,

Juancho.

Minimal example:

> from dqrobotics import *
> from dqrobotics.robots         import KukaLw4Robot
> from dqrobotics.robot_modeling import DQ_Kinematics
> import numpy as np
> 
> theta = np.random.rand(7,1)
> 
> robot = KukaLw4Robot.kinematics()
> x = robot.fkm(theta)   #Working
> print(x)
> 
> #robot.n_links    # AttributeError: 'dqrobotics.robot_modeling.DQ_SerialManipulator' object has no attribute 'n_links'
> robot.theta       # Works
> robot.d           # Works
> robot.a           # Works
> robot.alpha       # Works
> robot.dummy       # Works
> robot.n_dummy     # Works
> robot.convention  # Works
> robot.effector    # Works
> #robot.reference_frame  #AttributeError: 'dqrobotics.robot_modeling.DQ_SerialManipulator' object has no attribute 'reference_frame'
> #robot.base_frame     #AttributeError: 'dqrobotics.robot_modeling.DQ_SerialManipulator' object has no attribute 'base_frame'

mmmarinho, 2019-04-23:
Hello, @juanjqo,

It should work now. Note that `n_links` was replaced by `get_dim_configuration_space`
Also, we are going to remove dummy joints soon so `dummy', 'set_dummy', and 'n_dummy` are no longer available.

closed 2019-04-23 by mmmarinho

juanjqo, 2019-04-23:
@mmmarinho 
Thanks. However, now, I have another problem with those attributes.
Using the same minimal example, 
`print(robot.reference_frame)`

returns:
`<bound method PyCapsule.reference_frame of <dqrobotics.robot_modeling.DQ_SerialManipulator object at 0x7fb074b97f10>>`

Is this expected?
I was expecting a default dq as DQ([1,0,0,0,0,0,0,0])  (as in Matlab).

Cheers, 

Juancho

mmmarinho, 2019-04-23:
Hello!

It is a method not an attribute.
`print(robot.reference_frame())`
Should work.

juanjqo, 2019-04-23:
You're right!
Thank you!

commit 9a7a4fdb (2019-04-23): Updated cpp version, added base_frame, reference_frame, set_base_frame, set_reference_frame (Close #4)