dqrobotics/python #13: [REQUEST] is_line() function and others are not available in Python.

anachristinaac, 2019-10-23:
**Request description:**
Functions to check some properties of the DQ elements are not available in Python, such as *is_line()*, *is_plane()*, *is_pure()*, and *is_unit()*. However, they are available in Matlab. Is it possible to include them?

Other functions that are also available in Matlab but not in Python: *is_pure_quaternion()*, *is_quaternion()*, *is_real()*, *is_real_number()*.

**Minimal example in Matlab:**
*Code*
```
a  = DQ([1, 0, 0, 0, 0, 0, 0]);
is_line(a)
```

*Output*
```
ans =
	0
```

**Expected behavior in Python:**
I would expect the same behavior in Python. For instance:

*Code*
```
from dqrobotics import *
a = DQ([1, 0, 0, 0, 0, 0, 0])
is_line(a)
```

*Expected output:*
```
0
```

**Environment:**
OS Ubuntu 18.04
dqrobotics version 19.10.0.42
Python version 3.6.7
Matlab version R2018a

closed 2019-10-24 by mmmarinho

mmmarinho, 2019-10-24:
Hello, @anachristinaac.

Thanks for the report.

The following should work now:
```python
from dqrobotics import *

h = DQ([1, 0, 0, 0, 0, 0, 0, 0])

print('Is h unit', is_unit(h))
print('Is h pure?', is_pure(h))
print('Is h real?', is_real(h))
print('Is h a real number?', is_real_number(h))
print('Is h a quaternion?', is_quaternion(h))
print('Is h a pure quaternion?', is_pure_quaternion(h))
print('Is h a plane?', is_plane(h))
```

Output:
```
Is h unit? True
Is h pure? False
Is h real? True
Is h a real number? True
Is h a quaternion? True
Is h a pure quaternion? False
Is h a plane? True
```

anachristinaac, 2019-10-24:
Hi, @mmmarinho.
It's working. Thanks!

commit 2214e465 (2019-10-24): Fix #13