dqrobotics/python #14: [QUESTION] Was the DQ_LinearAlgebra already included in the stable version?

marcos-pereira, 2019-12-23:
Hello @mmmarinho,

I installed the DQ Robotics for python using `python3 -m pip install --user dqrobotics` and I also ran `python3 -m pip install --user dqrobotics --upgrade`. My current version is dqrobotics-19.10.0.53. 

How can I add the DQ_LinearAlgebra properly? If I import it with 
```
from dqrobotics.utils.DQ_LinearAlgebra import *
```

The library DQ_LinearAlgbera is not found and I get the following error when running the script with python3

```
Traceback (most recent call last):
  File "robot_controlling_node.py", line 3, in <module>
    from dqrobotics.utils.DQ_LinearAlgebra import *
ModuleNotFoundError: No module named 'dqrobotics.utils.DQ_LinearAlgebra'
```
The `dqrobotics` alone is found without errors.
Was the DQ_LinearAlgebra already included in the stable version?

My python3 version is 3.6.9 on Ubuntu 18.04.

Best regards,

Marcos

mmmarinho, 2019-12-25:
Hello, @marcos-pereira 

The way you tried importing is somewhat different from how it should be done in the current version of the library. 

To do what you want to do, you have to import the `utils` submodule.

```python
import numpy
from dqrobotics.utils import *

print(DQ_LinearAlgebra.pinv(numpy.identity(2)))
```

The output will be
```
[[1. 0.]
 [0. 1.]]
```

Basically, the command
```python
from X import *
```
on the current version will only work if `X` is a submodule (i.e. something similar to a folder in the cpp tree).

You can also add an alias for `pinv`
```python
import numpy
from dqrobotics.utils import *
pinv = DQ_LinearAlgebra.pinv

print(pinv(numpy.identity(2)))
```

I don't particularly like the current way this is done so I'll work on making the imports better for the next version of dqrobotics. Meanwhile, you can use what I described above.

closed 2019-12-25 by mmmarinho

marcos-pereira, 2019-12-25:
Thank you for the quick response @mmmarinho! Now it is working.

I would like just to point out that the python example [vrep_interface_move_kuka.py](https://github.com/dqrobotics/python-examples/blob/master/vrep_interface/vrep_interface_move_kuka.py) is using the `from dqrobotics.utils.DQ_LinearAlgebra import *
` command. I was using the example as a starting point.

Thanks, and happy holidays!

mmmarinho, 2019-12-28:
Hello, @marcos-pereira.

Nice to hear it's working. If there are any issues with the examples, please open a ticket so it's easier for us to keep track of what's odd (the issue tracker of dqrobotics/python is fine).

Happy holidays.

commit 1791bfea (2019-04-23): Updated cpp version, added base_frame, reference_frame, set_base_frame, set_reference_frame (Close #14)