dqrobotics/python #15: [QUESTION] Would it be possible to update the vrep_interface_move_kuka.py example?

marcos-pereira, 2019-12-28:
Hi @mmmarinho ,

As discussed on issue #14 , the example [vrep_interface_move_kuka.py](https://github.com/dqrobotics/python-examples/blob/master/vrep_interface/vrep_interface_move_kuka.py) is importing the DQ_LinearAlgebra in a wrong way which does not run.

Instead of using 
```
from dqrobotics.utils.DQ_LinearAlgebra import *
```
it should use
```
from dqrobotics.utils import *
```

Would it be possible to update the example?

Thanks,

Marcos

bvadorno, 2019-12-28:
Hi, @marcos-pereira.

Since the modification is simple and you already know how to do it, I suggest modifying it and then you can do a pull request. 

Best regards,
Bruno

marcos-pereira, 2019-12-28:
Hello @bvadorno ,

Thank you for the suggestion. I can do that. However, I found another issue related to importing `dqrobotics` submodules on the example [vrep_interface_move_kuka.py](https://github.com/dqrobotics/python-examples/blob/master/vrep_interface/vrep_interface_move_kuka.py). I opened issue #16 to solve the problem. Afterwards, I can fix both and do a pull request.

Best regards,
Marcos