dqrobotics/python #16: [QUESTION] How should one import the VrepInterface?

marcos-pereira, 2019-12-28:
Hello,

I found a similar problem to issue #14 when trying to import the VrepInterface in python.
I am currently trying to run the example [vrep_interface_move_kuka.py](https://github.com/dqrobotics/python-examples/blob/master/vrep_interface/vrep_interface_move_kuka.py). 

When the VrepInterface is imported with
```
from dqrobotics.interfaces import VrepInterface
```
a similar error to issue #14 occurs:
```
Traceback (most recent call last):
  File "vrep_interface_move_kuka.py", line 2, in <module>
    from dqrobotics.interfaces import VrepInterface
ImportError: cannot import name 'VrepInterface'
```
I tried the same approach from issue #14 but it didn't work. 

The DQ Robotics version is the dqrobotics-19.10.0.53. I have also installed the [V-REP Interface package](https://dqroboticsgithubio.readthedocs.io/en/latest/installation/cpp.html#interface-packages) .

Any hints on how to solve this one?

mmmarinho, 2019-12-29:
Hello, @marcos-pereira,

The correct class name is `DQ_VrepInterface` and the submodule is `vrep`.

```python
from dqrobotics.interfaces.vrep import DQ_VrepInterface
```

Let me know if it doesn't work well.

closed 2019-12-29 by mmmarinho

marcos-pereira, 2019-12-29:
Hi @mmmarinho ,

The DQ_VrepInterface was found, but it seems that the method `get_object_handles()` is not being found. I am probably missing some tiny detail.

I tried the following approach within the example [vrep_interface_move_kuka.py](https://github.com/dqrobotics/python-examples/blob/master/vrep_interface/vrep_interface_move_kuka.py)
```
from dqrobotics.interfaces.vrep import DQ_VrepInterface
...
## Creates a VrepInterface object
vi = DQ_VrepInterface()
...
## Connects to the localhost in port 19997 with timeout 100ms and 10 retries for each method call
vi.connect(19997,100,10)

## Starts simulation in V-REP
print("Starting V-REP simulation...")
vi.start_simulation()

## Store joint names
joint_names = ("LBR_iiwa_14_R820_joint1","LBR_iiwa_14_R820_joint2","LBR_iiwa_14_R820_joint3","LBR_iiwa_14_R820_joint4","LBR_iiwa_14_R820_joint5","LBR_iiwa_14_R820_joint6","LBR_iiwa_14_R820_joint7")
## Get joint handles
joint_handles = vi.get_object_handles(joint_names)
```
which has the following behavior
```
###############################################################
In order for this example to run correctly, you have to
open a scene in V-REP and drag a drop a "LBR_iiwa_14_R820" robot
and remove or disable its child script
before running this example.
###############################################################
Starting V-REP simulation...
'dqrobotics._dqrobotics._interfaces._vrep.DQ_VrepIn' object has no attribute 'get_object_handles'
There was an error connecting to V-REP, please check that it is open and that the Kuka Robot is in the scene.
```

I tested it with the newest CoppeliaSim version 4.0.0.

mmmarinho, 2019-12-29:
Hello! Thanks for trying to fix it.

That method does not exist anymore because it was superseded by better methods. In general, you should not use `get_object_handles`, hence why it was removed in the Python version. You can directly call most methods using the object's name, as a string.

See [DQ_VrepInterface.m](https://github.com/dqrobotics/matlab/blob/master/interfaces/DQ_VrepInterface.m) for the MATLAB class. It is well documented.

The following example might also be helpful
[vrep_interface_move_kuka_and_youbot.py](https://github.com/dqrobotics/python-examples/blob/master/vrep_interface/vrep_interface_move_kuka_and_youbot.py#L1)