dqrobotics/python #17: [QUESTION]

wdfalfred, 2020-03-09:
When I was trying to run the following code. 
//////////////////////////////////////////
import numpy as np
from math import *
from dqrobotics import *

phi_1 = pi/4         # set the rotation angel as pi/4
n_1 = DQ(0,1,0,0,0,0,0,0)     # set x as the rotation axis
p_1 = DQ(0,1,2,3,0,0,0,0)     # set the translation quaternion

phi_2 = -pi/4          # set the rotation angel as pi/4
n_2 = DQ(0,1,0,0,0,0,0,0)     # set x as the rotation axis
p_2 = DQ(0,1,2,3,0,0,0,0)     # set the translation quaternion

def udq_rigid_motion(phi, n, p):
    # calculate the rotation quaternion
    r = cos(phi/2) + sin(phi/2)*n
    # calculate the dual quaternion representing the rigid motion of translation p followed by rotation
    x = r + 0.5 * E_ * p *r
    print("udq_rigid_motion:")
    print(x)
    return x
x_01 = udq_rigid_motion(phi_1, n_1, p_1)
x_12 = udq_rigid_motion(phi_2, n_2, p_2)

x=x_01*x_12
# print( x.rotation() + 0.5 * E_ * x.translation() * x.rotation())
print(x.P())
print(2* x.D() * x.P().conj())
print(norm(x)== DQ(1,0,0,0,0,0,0,0))
print(translation(x))
//////////////////////////////////////////
The following error will show up. 
# Traceback (most recent call last):
#   File "/home/winford/PycharmProjects/mpc/venv/dqrobotics/dqrobot_test.py", line 52, in <module>
#     print(translation(x))
# ValueError: Bad translation() call: Not a unit dual quaternion
It says that x is not a unit dual quternion. I running the code "norm(x)" the result shows that 1 0i 0j 0k +E( 0 0i 0j 0k ), but "print(norm(x)" == DQ(1,0,0,0,0,0,0,0)) shows "False"

closed 2020-03-09 by mmmarinho

mmmarinho, 2020-03-09:
Hello, @wdfalfred,

Thank you for the question. I tested locally with the script bellow and could replicate your issue. 
  
```python
from dqrobotics import *
from math import *
import numpy as np

phi = pi / 4.0  # Set the rotation angle
n = DQ(0, 1, 0, 0, 0, 0, 0, 0)  # Set the rotation axis
t = DQ(0, 1, 2, 3, 0, 0, 0, 0)  # Set the translation

r = cos(phi / 2.0) + n * sin(phi / 2.0) # Get rotation quaternion
x = r + 0.5 * E_ * t* r # Get dual quaternion pose

np.set_printoptions(precision=12)
print("x = {}".format(x.q))
print("is_unit(x) = {}".format(is_unit(x)))
print("translation(x) = {}".format(translation(x)))
```

I was able to find the source of the error and fixed it in commit 068bbaf. Please update your DQ Robotics Python to version 19.10.1.1 and let me know if you still have any issues.

Kind regards,
Murilo

wdfalfred, 2020-03-09:
Hello Marinho, 
Thank you for your fast response. I have solved the problem by following your instructions.
Best regards,
Daifeng Wang

commit 068bbafd (2020-03-09): Fix #17