dqrobotics/python #18: Unexpected behavior using rotation_axis() method [BUG]

juanjqo, 2020-07-06:
Hi @mmmarinho! I am having some problems using the rotation_axis() method. Would you give me a hand?
I appreciate your support.

**Bug description**
- The method rotation_axis() returns a wrong result. In addition, when I use print(r), the value is not printed properly.

*Code*

```
from dqrobotics import*

r = DQ([0.5068154, -0.8591303, 0.0555768, -0.0440969])
r = normalize(r)
print('r:', r)
print('axis: ', r.rotation_axis())
print('angle: ', r.rotation_angle())

```
*Output*
```
r: 0
axis:  0
angle:  2.0786195489067434

Process finished with exit code 0
```

**Expected behavior**
- I expect the right value for the rotation when I use print(r), and the right rotation axis when I use rotation_axis(). The same code in Matlab is working.

*Expected output (Matlab)*
```
clear all
close all
clc

r = DQ([0.5068154, -0.8591303, 0.0555768, -0.0440969]);

r = normalize(r);

r
r.rotation_axis()
r.rotation_angle()
```

*Matlab Output*
```
r = 
         0.50682 - 0.85913i + 0.055577j - 0.044097k
ans = 
          - 0.99661i + 0.06447j - 0.051153k

ans =

    2.0786
```

**Environment:**
 - OS: Ubuntu 18.04.4 LTS
 - dqrobotics version 20.4.0.10 (Python)
 - Python version 3.6
 - MATLAB version 2018b

mmmarinho, 2020-07-07:
Hello, @juanjqo 

If you test the following code using a debugger, you'll see that the axis and angle are being calculated correctly.
```python
r = DQ([0.5068154, -0.8591303, 0.0555768, -0.0440969])
r = normalize(r)
axis = r.rotation_axis()
angle = r.rotation_angle()
```

However, the print function is indeed malfunctioning. I'll check it and come back to you.

juanjqo, 2020-07-08:
@mmmarinho Thanks for your answer!

Yes, indeed, the axis and angle are being calculated correctly. My bad, the bug is related with the printing function.

I realize that you closed the issue (Unexpected behavior when printing DQs [BUG]#18). I updated to the last version: 20.4.0.11 However, the printing bug is still present.

```
from dqrobotics import*
r = DQ([0.5068154, -0.8591303, 0.0555768, -0.0440969])
r = normalize(r)
axis = r.rotation_axis()
angle = r.rotation_angle()

print('r', vec4(r))
print('axis: ', vec4(axis))
print('angle: ', angle)

print('r', r)
print('axis: ', axis)
```
Output:
```
r [ 0.50681539 -0.85913028  0.0555768  -0.0440969 ]
axis:  [ 0.         -0.99660772  0.06447016 -0.05115325]
angle:  2.0786195489067434
r 0
axis:  0

Process finished with exit code 0
```

Cheers!

mmmarinho, 2020-07-09:
@juanjqo 

Thank you for the message. Locally, it is working. I'll try to figure out what is wrong.

mmmarinho, 2020-07-09:
Thanks again for the report, @juanjqo.

Fixed in dqrobotics/cpp@dd72ae8, available in Python since d5595266.
Please update to `dqrobotics-20.4.0.18` and let me know if something malfunctioning.

closed 2020-07-09 by mmmarinho

juanjqo, 2020-07-09:
@mmmarinho Thank you!
 I updated to the last version (dqrobotics-20.4.0.17) using:
`python3 -m pip install --user dqrobotics`
 It is working now!  :D

commit 7437c7c1 (2020-07-07): Adding to_string(). Updating with cpp to fix #18.