dqrobotics/python #21: [BUG] python3 dqrobotics is not upgrading to 20.4.0.21

marcos-pereira, 2020-11-05:
**Bug description**
- The dqrobotics 20.4.0.21 is not being installed when I ask dqrobotics to upgrade although it is found with `pip search`.

**To Reproduce**

*Code*

```
python3 -m pip install --user dqrobotics --upgrade
```
*Output*
```
Requirement already up-to-date: dqrobotics in /home/marcos/.local/lib/python3.6/site-packages (20.4.0.17)
Requirement already satisfied, skipping upgrade: numpy in /home/marcos/.local/lib/python3.6/site-packages (from dqrobotics) (1.19.4)
```

**Expected behavior**
- dqrobotics 20.4.0.21 should be installed

**Environment:**
 - OS: Ubuntu 18.04
 - dqrobotics version 20.4.0.17
 - Python version 3.6.9

**Additional context**
If I specify the version to be installed, 20.4.0.21 is not found:
```
pip install dqrobotics==20.4.0.21
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement dqrobotics==20.4.0.21 (from versions: 0.0.166, 0.0.168, 0.0.170, 0.1.0.38, 0.1.0.39, 0.1.0.40, 0.1.0.42, 0.1.0.43, 0.1.0.44, 0.1.0.45, 0.1.0.46, 0.1.0.47, 0.1.0.48, 0.1.0.49, 0.1.0.50, 0.1.0.51, 0.1.0.52, 0.1.1, 0.1.1.1, 0.1.1.2, 0.1.1.3, 0.1.1.4, 0.1.1.5, 0.1.1.8, 0.1.1.9, 0.1.1.10, 0.1.1.11, 0.1.1.12, 0.1.1.15, 0.1.1.18, 0.1.1.19, 0.1.1.20, 0.1.1.21, 0.1.1.22, 0.1.1.27, 0.1.1.28, 0.1.1.29, 0.1.1.30, 0.1.1.37, 0.1.1.38, 0.1.1.39, 0.1.1.40, 0.1.1.42, 0.1.1.43, 0.1.1.44, 0.1.1.45, 0.1.1.46, 0.1.1.47, 0.1.1.48, 0.1.1.51, 0.1.1.52, 0.1.2.1, 0.1.2.2, 0.1.2.3, 0.1.2.4, 0.1.2.5, 0.1.2.6, 0.1.2.7, 0.1.2.8, 0.1.2.9, 0.1.2.10, 0.1.2.11, 0.1.2.12, 0.1.2.13, 0.1.2.14, 0.1.2.15, 0.1.2.16, 0.1.2.17, 0.1.2.18, 0.1.2.19, 0.1.2.20, 0.1.2.21, 0.1.2.23, 0.1.2.24, 0.1.2.25, 0.1.2.26, 0.1.2.27, 0.1.2.28, 0.1.2.29, 0.1.2.34, 0.1.2.35, 0.1.2.36, 19.10.0.0, 19.10.0.1, 19.10.0.2, 19.10.0.3, 19.10.0.4, 19.10.0.16, 19.10.0.18, 19.10.0.20, 19.10.0.21, 19.10.0.22, 19.10.0.26, 19.10.0.27, 19.10.0.29, 19.10.0.30, 19.10.0.32, 19.10.0.33, 19.10.0.34, 19.10.0.35, 19.10.0.36, 19.10.0.37, 19.10.0.38, 19.10.0.39, 19.10.0.42, 19.10.0.43, 19.10.0.44, 19.10.0.45, 19.10.0.46, 19.10.0.47, 19.10.0.48, 19.10.0.49, 19.10.0.50, 19.10.0.51, 19.10.0.52, 19.10.0.53, 19.10.0.54, 19.10.0.55, 19.10.0.57, 19.10.1.0, 19.10.1.1, 19.10.1.2, 19.10.1.3, 19.10.1.5, 19.10.1.6, 19.10.1.7, 19.10.1.8, 19.10.1.9, 19.10.1.10, 19.10.1.11, 19.10.1.12, 19.10.1.13, 19.10.1.14, 19.10.1.15, 19.10.1.27, 19.10.1.28, 19.10.1.29, 19.10.1.38, 20.4.0.1, 20.4.0.4, 20.4.0.5, 20.4.0.6, 20.4.0.7, 20.4.0.8, 20.4.0.9, 20.4.0.10, 20.4.0.11, 20.4.0.12, 20.4.0.13, 20.4.0.17)
ERROR: No matching distribution found for dqrobotics==20.4.0.21
```
I also tried downloading the [file .whl](https://files.pythonhosted.org/packages/f5/61/236dcd1dec06b9db4cba793788ac08474bf748df8e1f4f3ce3abf0430609/dqrobotics-20.4.0.21-cp35-cp35m-manylinux1_x86_64.whl) from [Pypi](https://pypi.org/project/dqrobotics/20.4.0.21/#files) and install, but without success:
```
pip3 install dqrobotics-20.4.0.21-cp35-cp35m-manylinux1_x86_64.whl 
Defaulting to user installation because normal site-packages is not writeable
ERROR: dqrobotics-20.4.0.21-cp35-cp35m-manylinux1_x86_64.whl is not a supported wheel on this platform.
```
Finally, if I run pip search:
```
pip search dqrobotics
dqrobotics (20.4.0.21)  - DQRobotics python
  INSTALLED: 20.4.0.17
  LATEST:    20.4.0.21
```

Best regards,
Marcos

mmmarinho, 2020-11-05:
Hello, @marcos-pereira,

Thanks for the report. It should be working now but let me know if it doesn’t.

Fixed in 8b17a9a.

Cheers,
Murilo

closed 2020-11-05 by mmmarinho

marcos-pereira, 2020-11-05:
Thanks, @mmmarinho ! It is working now.