dqrobotics/python #25: [BUG] The available robots models are not working with the DQ_QuadraticProgrammingSolver

marcos-pereira, 2021-03-25:
**Bug description**
- The available robots models are not working with the DQ_QuadraticProgrammingSolver.

**To Reproduce**

*Code*

```
from dqrobotics import *
from dqrobotics.solvers import DQ_QuadraticProgrammingSolver
import dqrobotics.robot_control as dq_controllers
from dqrobotics.robot_control import ControlObjective
from dqrobotics.robots           import KukaLw4Robot

robot = KukaLw4Robot.kinematics()
qp_solver = DQ_QuadraticProgrammingSolver
controller = dq_controllers.DQ_ClassicQPController(robot, qp_solver)
controller.set_control_objective(ControlObjective.Pose)
```
*Output*
```
Traceback (most recent call last):
  File "qp_solver_issue.py", line 9, in <module>
    controller = dq_controllers.DQ_ClassicQPController(robot, qp_solver)
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. dqrobotics._dqrobotics._robot_control.DQ_ClassicQPController(arg0: dqrobotics._dqrobotics._robot_modeling.DQ_Kinematics, arg1: dqrobotics._dqrobotics._solvers.DQ_QuadraticProgrammingSolver)

Invoked with: <dqrobotics._dqrobotics._robot_modeling.DQ_SerialManipulatorDH object at 0x7f3b34cfa3f0>, <class 'dqrobotics._dqrobotics._solvers.DQ_QuadraticProgrammingSolver'>
```

**Expected behavior**
- The DQ_QuadraticProgrammingSolver should be created without errors. 

**Environment:**
 - OS: Ubuntu 20.04
 - dqrobotics version: 20.4.0.27
 - Python version 3.8.5
 
**Additional information**
- I believe there was some change in the DQ_QuadraticProgrammingSolver that used to accept KukaLw4Robot.kinematics(), but now it requires the newer DQ_SerialManipulatorDH. However, the KukaLw4Robot was not updated. 
- If this is the case, it would also be interesting to know the right way to create our own robots objects in Python. I was using the following code to create a robot:
```
from dqrobotics import *
from dqrobotics.robot_modeling import DQ_SerialManipulatorDH
import numpy as np
from math import pi
pi2 = pi/2
def KukaLBR4pRobot():
    kuka_lbr4p_theta = np.array([0,     0,     0,   0,   0,    0,   0])
    kuka_lbr4p_d = np.array([0.2,     0,     0.4, 0,   0.39, 0,   0])
    kuka_lbr4p_a = np.array([0,     0,     0,   0,   0,    0,   0])
    kuka_lbr4p_alpha = np.array([pi2,   -pi2,  pi2, -pi2, pi2, -pi2, 0])
    kuka_lbr4p_type = np.array([0, 0, 0, 0, 0, 0, 0])
    kuka_DH_matrix = np.array([kuka_lbr4p_theta,
                               kuka_lbr4p_d,
                               kuka_lbr4p_a,
                               kuka_lbr4p_alpha,
                               kuka_lbr4p_type])
    kuka_lbr4p = DQ_SerialManipulatorDH(kuka_DH_matrix, 'standard')
    return kuka_lbr4p
```

mmmarinho, 2021-03-28:
@marcos-pereira 

Thanks for the report. 

I think this bug is not present when using dqrobotics/cpp#34. However, it might take us some time to update it.

marcos-pereira, 2021-03-28:
@mmmarinho , thanks for the answer. In the meantime, I will migrate some of my python codes to cplusplus.

mmmarinho, 2021-03-28:
@marcos-pereira 

Sorry, but there was actually no bug. The minimal example you sent is incorrect because `DQ_QuadraticProgrammingSolver` is an abstract class and cannot be instanced (it is not supposed to be).

You need to choose a concrete subclass of `DQ_QuadraticProgrammingSolver` instead. In Python, I recommend starting with `DQ_QuadProgSolver`

```python
from dqrobotics import *
from dqrobotics.solvers import DQ_QuadprogSolver
import dqrobotics.robot_control as dq_controllers
from dqrobotics.robot_control import ControlObjective
from dqrobotics.robots import KukaLw4Robot

robot = KukaLw4Robot.kinematics()
qp_solver = DQ_QuadprogSolver()
controller = dq_controllers.DQ_ClassicQPController(robot, qp_solver)
controller.set_control_objective(ControlObjective.Pose)
```

To use it, don't forget to install `quadprog`.
`python3 -m pip install quadprog --user`

closed 2021-03-28 by mmmarinho

marcos-pereira, 2021-04-06:
@mmmarinho many thanks! 
I was forgetting to install quadprog. 
Maybe you could add this information to the ReadTheDocs Python installation. Similar to the C++ interfaces where you mention the need for [CPLEX](https://dqroboticsgithubio.readthedocs.io/en/latest/installation/cpp.html#interface-packages).

mmmarinho, 2021-04-06:
@marcos-pereira 

Indeed. In fact, you're welcome to update the docs with that information and generate a pull request.