dqrobotics/python #39: [QUESTION] M1 Mac pip

JPJSea, 2022-10-13:
Previously when executing
```
pip install dqrobotics
```
on an M1 Mac (Big Sur) I have received an error message
```
ERROR: Could not find a version that satisfies the requirement dqrobotics (from versions: none)
ERROR: No matching distribution found for dqrobotics
```
This is fine. I just:
```
python -m pip install 'dqrobotics @ git+https://github.com/dqrobotics/python.git'
```
and everything was ok, no problems using the library since a couple of months now.

Fast forward to today in a fresh venv I was surprised to see that the library downloads and installs with `pip install dqrobotics`, however when I exectute the script I have wrong arch errors:
```
ImportError: dlopen(/Users/user/Documents/envs/test/lib/python3.10/site-packages/dqrobotics/_dqrobotics.cpython-310-darwin.so, 2): no suitable image found.  Did find:
	/Users/user/Documents/envs/test/lib/python3.10/site-packages/dqrobotics/_dqrobotics.cpython-310-darwin.so: mach-o, but wrong architecture
	/Users/user/Documents/envs/test/lib/python3.10/site-packages/dqrobotics/_dqrobotics.cpython-310-darwin.so: mach-o, but wrong architecture
```
Again, I'm fine just installing the library from the repo, everything works correctly, but my question is, are M1 Macs intended to be supported?

mmmarinho, 2022-10-13:
@JPJSea Thank you for the question. I was not able to reproduce your error, so let's see if we can pinpoint what is going on in your system.

To reply directly to your question, DQRobotics Python is supposed to be supported *as long as it works on my Mac M1*, because the binaries on PyPI are built on my Mac. This means no exoteric installations of Python (I use whatever is default or available via homebrew) and the latest macOS.

That said, I think the problem you're having is that you are not trying to install the pre-release version, which is the one that actually builds for the M1. You'll need to use the `--pre` flag until the `master` updates are pushed into the `release` branch (@bvadorno). This is intended, not a documentation issue, and PyPI is supposed to work like that.

Does this work?
```commandLine
pip install dqrobotics --pre
```

JPJSea, 2022-10-15:
I set up a new env, purged the pip cache and updated pip (just in case) and used the `--pre` flag, this produced the same import error. This is it with with the traceback:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/user/deb/lib/python3.10/site-packages/dqrobotics/__init__.py", line 26, in <module>
    from dqrobotics._dqrobotics import *
ImportError: dlopen(/Users/user/deb/lib/python3.10/site-packages/dqrobotics/_dqrobotics.cpython-310-darwin.so, 2): no suitable image found.  Did find:
	/Users/user/deb/lib/python3.10/site-packages/dqrobotics/_dqrobotics.cpython-310-darwin.so: mach-o, but wrong architecture
	/Users/user/deb/lib/python3.10/site-packages/dqrobotics/_dqrobotics.cpython-310-darwin.so: mach-o, but wrong architecture
```

As I said I am on Big Sur (I don't know if that is the issue). Python is v3.10.8 installed with homebrew.

mmmarinho, 2022-10-15:
@JPJSea 

You first need to remove that old installation of `dqrobotics`, because you're still using it.
I can see from the shared object name
`_dqrobotics.cpython-310-darwin.so`
which does not exist in PyPI. See [here](https://pypi.org/project/dqrobotics/21.4.0a58/#files).

Supposing that you made your Python 3.10 default in your terminal, this should suffice
`python -m pip uninstall dqrobotics`
or change to the correct alias to achieve this. Remember: you have to do this outside your venv because this installation was made outside your venv. I can tell from the path
`/Users/user/deb/lib/python3.10/site-packages`

After doing the above *correctly* and reconfiguring your venv, there are only two possible outcomes. (Infer from this that if there's any other output, that old installation still lingers)

1. It works when using `--pre`
2. `pip` complains again with 
```
ERROR: Could not find a version that satisfies the requirement dqrobotics (from versions: none)
ERROR: No matching distribution found for dqrobotics
```

In case of 2., if `pip` still complains even with the `--pre` flag, then this most likely means that either you'll have to upgrade to Monterey or try to build it locally. 

Let me know how it goes and I can try to guide you through building it locally if you need. It's not difficult.

JPJSea, 2022-10-16:
> @JPJSea
> 
> You first need to remove that old installation of `dqrobotics`, because you're still using it. I can see from the shared object name `_dqrobotics.cpython-310-darwin.so` which does not exist in PyPI. See [here](https://pypi.org/project/dqrobotics/21.4.0a58/#files).
> 

I have never installed this library outside of a venv. There is no installation of it anywhere on my machine other than a virtual env, the path to the object is inside the enve `/Users/user/deb/lib`. The name of the env is `deb` an abbreviation of debug. I can confirm this with `find /Users -name "dqrobotics"`.  It shows only this env and Matlab toolbox paths.

Evidently if I try to uninstall the library outside of this venv, pip says:
```
WARNING: Skipping dqrobotics as it is not installed.
``` 

I already said that I have no problems building locally and have been using this library fine after doing so. This is why I posed this as a question and not a bug. Feel free to close this issue, as my original question has been answered.

mmmarinho, 2022-10-16:
As you wish.

closed 2022-10-16 by mmmarinho