dqrobotics/python #49: [BUG] Unknown type `DQ_robotics::SOMETHING`

mmmarinho, 2023-07-06:
**Bug description**

This is an issue to help me start tracking this bug as I haven't been able to fully identify when it happens.

**To Reproduce**

- Have a version of `dqrobotics` Python3 installed, let's call it XXX.
- Try to make that work with another pybind11 compiled with a (slightly different) version of `dqrobotics`, let's call it YYY.

**Problem**

All sorts of bugs involving `Unknown type` and the `DQ_robotics` namespace.

**Solution**

The Python3 version of `dqrobotics` can be fixed by being compiled locally. This mismatch might be due to the environment used to compile with GitHub actions, that might need adjustments (I'm looking at you, GCC version). 

Remember to source your `venv` if you're using one
```console
python3 -m pip uninstall dqrobotics
git clone https://github.com/dqrobotics/python.git dqrobotics_python --recursive
python3 -m pip install wheel scipy quadprog
python3 -m pip install ./dqrobotics_python 

cd dqrobotics_python/tests
python DQ_test.py
python DQ_Kinematics_test.py
python cpp_issues.py
python python_issues.py
cd ..
```

bvadorno, 2023-10-29:
@mmmarinho, any new developments on this?

mmmarinho, 2023-10-29:
@bvadorno 

The major cause for this was originally addressed in https://github.com/dqrobotics/python/issues/44. That fix should solve most occurrences caused by our choices and I also mention how to maintain it for future versions.

However, there are near infinite situations in which this import error can happen and most of them related to complex interactions with the user environment and advanced use of the library with custom Python bindings and custom C++ code. At this stage, I don't have enough information to propose a universal fix (beyond compiling locally as described above), but I'm sure this issue will resurface for advanced users. 

When it does resurface, I'd prefer to center all the related discussion in this issue instead of having it spread between smaller, seemingly disjoint issues, when in fact they have the same underlying cause.

qlin960618, 2025-06-08:
Hi @mmmarinho, 

Thanks in advance for your assistance!
For some reason I am not sure why this problem is resurfacing again... specificly in something that I am doing.

Following back to the original **concrete example** in [tro2022_adaptivecontrol](https://github.com/mmmarinho/tro2022_adaptivecontrol)

```shell
./test_python_wrapper.sh 
```
OUTPUT 
```
  File "???/site-packages/marinholab/papers/tro2022/adaptive_control/__init__.py", line 3, in <module>
    from marinholab.papers.tro2022.adaptive_control._core import *
ImportError: generic_type: type "Example_SerialManipulatorEDH" referenced unknown base type "DQ_robotics::DQ_SerialManipulator"
```
Something else I tried changing in test_python_wrapper.sh 
```shell
#python3 -m pip install dqrobotics --pre
python3 -m pip install ./python_wrapper/dqrobotics_python
```
in the original script using system env or venv by adding 
```
python3.12 -m venv test_env 
source test_env/bin/activate
```
Is there any pointer in how I can solve this issue? I notice there are several commit there recently attempting to solve the same problem?

this is my current setup for reference
MacOS: Sequoia 15.5 (24F74)
Python version: Python 3.12.11 
```shell
whereis python3.12
python3.12: /opt/homebrew/bin/python3.12 /opt/homebrew/share/man/man1/python3.12.1
```

mmmarinho, 2025-06-08:
Hi @qlin960618,

Whenever you have this problem it’s always because the packages you have installed are out of sync. Sometimes a problem on your end, sometimes on my end. 

Could you please start by uninstalling the packages (possibly cleaning the cache). Check at PyPi if you really have the latest ones. 
https://pypi.org/project/dqrobotics/#history

Then reinstall starting from `DQRobotics` with `—pre`?

If you’re sure you have the latest packages then I’ll check it on my end.

qlin960618, 2025-06-08:
Hi @mmmarinho 
I can confirm that I have clear the cache with 
```shell
python3.12 -m pip cache purge
```
aside from that, I am still getting the same error. 
with 
```python3 -m pip install dqrobotics --pre```
Which I am getting *25.4.0a7* for the latest
or
```python3 -m pip install ./python_wrapper/dqrobotics_python```

The other recent things I change is upgrading my MacOS, but I have no way of reverting that I think...

mmmarinho, 2025-06-08:
@qlin960618 

This worked for me in MacOS just now

Replace the `python3.11` with whatever python you want to use. Please use `3.10` and above.

```
cd ~
mkdir testissue
cd testissue
python3 -m venv venv
source venv/bin/activate
python -m pip install -U pip
python3 -m pip install dqrobotics --pre
pip install marinholab-papers-tro2022-adaptivecontrol
```

Then try

```
python
Python 3.11.12 (main, Apr  8 2025, 14:15:29) [Clang 17.0.0 (clang-1700.0.13.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from marinholab.papers.tro2022.adaptive_control import *
```

It imported with no errors for me.

qlin960618, 2025-06-09:
Hi @mmmarinho 

I can confirm pulling the package from pypi works without problem here....

So issue only exist if I try compiling locally???? Not sure what is the problem here. 

I ve replace the test_python_wrapper.sh with the following
```shell
python3.12 -m venv test_venv
source test_venv/bin/activate

#python3 -m pip config set global.break-system-packages true
python3.12 -m pip install --upgrade pip
python3.12 -m pip install setuptools wheel setuptools-git-versioning

#python3.12 -m pip install dqrobotics --pre
python3.12 -m pip install ./python_wrapper/dqrobotics_python/.
python3.12 -m pip install ./python_wrapper
#python ./python_wrapper/setup.py build_ext --verbose

cat adaptive_control_import_eval.py
python3.12 adaptive_control_import_eval.py
```

just for reference this is my gcc version
```shell
 gcc --version
Apple clang version 17.0.0 (clang-1700.0.13.5)
Target: arm64-apple-darwin24.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
and cmake 
```shell
cmake version 4.0.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
I have also tried reinstall cmake with ```brew reinstall``` and nuking the Xcode into trash with still no dice....

mmmarinho, 2025-06-09:
Hi @qlin960618,

Thank you for the enlivened comment.

I suppose that with the local compilation it does not work for you yet. This is one way that should work.
(The train's internet isn't the best so I'm 90% sure this works)

```
cd ~
mkdir testissueagain
cd testissueagain

git clone https://github.com/mmmarinho/tro2022_adaptivecontrol.git --recursive
cd tro2022_adaptivecontrol.git

python3.12 -m venv test_venv
source test_venv/bin/activate

python -m pip install --upgrade pip
python -m pip install setuptools wheel setuptools-git-versioning

python -m pip install ./python_wrapper/dqrobotics_python
python -m pip install ./python_wrapper
```

mmmarinho, 2025-06-09:
Hi @qlin960618 

The main difference is 

```
git clone https://github.com/mmmarinho/tro2022_adaptivecontrol.git --recursive
```

so that you get a fresh clone with the latest code. This is a way to mimic what's done in the continuous integration that creates the packages for PyPI.

qlin960618, 2025-06-09:
Hi @mmmarinho 

thanks! ic...
for the run that I did, i did  use `--recursive`. So maybe that is not the cause?

mmmarinho, 2025-06-09:
Right, thanks for letting me know @qlin960618.

If everything is built in the same system in an organized way it will always work. It will be more difficult if you try to have some packages locally built and some from PyPI, for instance. 

I’m closing this issue because it’s not a problem in this repo but if you need a hand I can help you through other channels.