dqrobotics/python #52: [ENHANCEMENT] GIthub hosted apple silicon runners

mmmarinho, 2023-10-03:
The current GitHub Actions workflow is using a self-hosted M1 runner to build the Python package for that enviroment. There seems to be no currently available methods to use the github-hosted Apple silicon machines free for open-source packages.

However, with the recent release of the so-called large Apple silicon machines (not free), we can see a glimpse of hope for github-hosted Apple silicon runners.

This issue is to keep track of this enhancement proposal and, in due time, replace the self-hosted machine by the github-hosted one.

https://github.com/actions/runner-images/issues/8439
https://github.blog/2023-10-02-introducing-the-new-apple-silicon-powered-m1-macos-larger-runner-for-github-actions/

radeva, 2024-01-18:
@mmmarinho You can try [FlyCI](https://www.flyci.net/)'s M1 and M2 runners. They are on average 2x faster and 2x cheaper than GitHub's **AND** we have a **free tier for OSS projects** (see below).

# Install Instructions

1. Install [FlyCI app](https://github.com/apps/flyci-prod) and 
2. Easily replace one line of code and start using FlyCI runners:

```diff
jobs:
 ci:
-    runs-on: macos-latest
+    runs-on: flyci-macos-large-latest-m1
   steps:
   - name: 👀 Checkout repo
     uses: actions/checkout@v4
```
# 500 mins/month Free for Public Repos

Since your repo is public, FlyCI offers **500 mins/month of free M1 runner usage** with the `flyci-macos-large-latest-m1` runner for public projects. 

Don't hesitate to contact us in case the free tier doesn't suit your needs or you experience any issues with the runners. Our team is here to support you!

Best Regards,
Veselina Radeva
Product Manager at [FlyCI](https://flyci.net/)

mmmarinho, 2024-07-12:
Solved in 363627c. See #58.

closed 2024-07-12 by mmmarinho