dqrobotics/python #53: [BUG]

marcos-pereira, 2023-11-01:
**************************
### Code of Conduct
_By submitting this report you automatically agree that you've read and accepted the following conditions._ 
- Support for DQ Robotics is given voluntarily and it's not the developers' role to educate and/or convince anyone of their vision.
- Any possible response and its timeliness will be based on the relevance, accuracy, and politeness of a request and the following discussion.
- If a DQ Robotics member replies, the user must let them know if their response solves their issue or not. 
- Any suggestion/advice/request made by anyone, as well intentioned as they might be, might not be incorporated into DQ Robotics.
*************************

**Bug description**
The DQ_VrepInterface is not connecting to the CoppeliaSim. 

**To Reproduce**
Open an empty CoppeliSim scene and run the code below.

*Code*

```python
from dqrobotics.interfaces.vrep import DQ_VrepInterface
import time
# This port needs to be configured correctly in your CoppeliaSim!
remote_api_port = 19997 
    
vi = DQ_VrepInterface()    

## Connects to the localhost in port 19997 with timeout 100ms and 10 retries for each method call
vi.connect("127.0.0.1",remote_api_port, 100, 10)    

if not vi.connect("127.0.0.1", remote_api_port, 100, 10):
    raise RuntimeError("Unable to connect to CoppeliaSim, be sure CoppeliaSim is opened in the default scene "
                    "and that port {} is correctly opened.".format(remote_api_port))

## Starts simulation in V-REP
print("Starting V-REP simulation...")
vi.start_simulation()
time.sleep(1) 
vi.stop_simulation()

vi.disconnect_all()
```

*Output*
```
line 12, in <module>
    raise RuntimeError("Unable to connect to CoppeliaSim, be sure CoppeliaSim is opened in the default scene "
RuntimeError: Unable to connect to CoppeliaSim, be sure CoppeliaSim is opened in the default scene and that port 19997 is correctly opened.
```

**Expected behavior**
The simulation scene should start, run for 1 second and stop.

*Expected output*
```
Starting V-REP simulation...
```

**Environment:**
 - OS: Ubuntu 22.04
 - dqrobotics version 23.4.0a18
 - Python version 3.10.12
 - CoppeliaSim version: 4.6.0-rev0
 - libdqrobotics-interface-vrep: 19.10.0-0~202310100528~ubuntu22.04.1

**Additional context**
- The DQ_VrepInterface is working properly on Matlab. It worked after adding the legacy remote API folder from the CoppeliaSim to the Matlab path `coppeliasim-4-6-0-rev0/programming/legacyRemoteApi/remoteApiBindings/matlab/matlab`. I also had to add `coppeliasim-4-6-0-rev0/programming/legacyRemoteApi/remoteApiBindings/lib/lib/Ubuntu20_04` to the path.
- Matlab version: R2023a
- I reported the bug directly to @juanjqo before and he asked me to open the issue for us to discuss.

mmmarinho, 2023-11-01:
@marcos-pereira Thanks. Version 4.6.0 is the one released about 10 days ago? 

This type of 0-ish-day compatibility is not something I can keep up with (nor want to), but maybe can be considered if someone can shoulder that responsibility.

The Python version of VrepInterface at `—pre` is currently meant to be compatible with `coppeliasim-v4.2.0-rev5`.

If you need a newer version, I recommend using the CPP library while we streamline interface versioning for Python as well, but let me assure you that this will not be easy nor fast. 

For the CPP version, I think it’s compiling against 4.5.1. My usual advice is to never target the latest version of CoppeliaSim unless you have a very good reason for it, usually some sorely needed new capability.

juanjqo, 2023-11-01:
Hi @marcos-pereira, 

As mentioned by @mmmarinho, the Python version (--pre) is compatible with `coppeliasim-v4.2.0-rev5`. Still, I think that could work with the last version of CoppeliaSim, at least for most of the functions.

I checked your code, and it looks like you are trying to connect to the same scene two times at the same port. If you comment line 11 (i.e., `vi.connect("127.0.0.1", remote_api_port, 100, 10`) your code should work.

Code after the modification:

```python
# This is a sample Python script.
from dqrobotics.interfaces.vrep import DQ_VrepInterface
import time

# This port needs to be configured correctly in your CoppeliaSim!
remote_api_port = 19997

vi = DQ_VrepInterface()

## Connects to the localhost in port 19997 with timeout 100ms and 10 retries for each method call
#vi.connect("127.0.0.1", remote_api_port, 100, 10)

if not vi.connect("127.0.0.1", remote_api_port, 100, 10):
    raise RuntimeError("Unable to connect to CoppeliaSim, be sure CoppeliaSim is opened in the default scene "
                       "and that port {} is correctly opened.".format(remote_api_port))

## Starts simulation in V-REP
print("Starting V-REP simulation...")
vi.start_simulation()
time.sleep(1)
vi.stop_simulation()

vi.disconnect_all()
```

By the way, I think you could improve your example using some [good practices](https://ros2-tutorial.readthedocs.io/en/latest/preamble/python/python_best_practices.html)

Minimal example:

```python
#!/bin/python3
from dqrobotics.interfaces.vrep import DQ_VrepInterface
import time

vi = DQ_VrepInterface()

def main() -> None:
    vi = DQ_VrepInterface()
    try:
        if not vi.connect("127.0.0.1", 19997, 100, 10):
            raise RuntimeError("Unable to connect to CoppeliaSim.")

        vi.start_simulation()
        time.sleep(0.1)
        # --------------Your code here----------------------------

        # ---------------------------------------------------------

    except (Exception, KeyboardInterrupt) as e:
        print(e)
        pass

    finally:
        vi.stop_simulation()
        vi.disconnect()

if __name__ == "__main__":
    main()
```

Please let me know if the problem persists.

Best regards, 

Juancho

mmmarinho, 2023-11-01:
@juanjqo Thanks for the insightful advice for @marcos-pereira.

Despite this, I have mixed feelings about giving advice on unsupported configurations, because that’s a slippery slope. You know how the remoteAPI can fail in unpredictable ways when the versions don’t match.

I certainly see how @marcos-pereira can benefit from your advice, but, despite that, the takeaway message from these issues should always be that users should stick to the supported versions for development and error reporting.

juanjqo, 2023-11-01:
@mmmarinho I completely agree. I think my reply wasn't as I wanted it to be 😅.

@marcos-pereira as pointed out by Murilo, I would recommend you to use only the supported versions. On the other hand, I believe that the example you provided will fail in any version of CoppeliaSim because you are trying to connect two times at the same scene using the same port.

marcos-pereira, 2023-11-01:
Hi @juanjqo @mmmarinho ,

Thanks for the quick reply.

I tried connecting only once in my example by removing the repeated call to connect and still did not work.

I tested the best practice example from @juanjqo and it worked correctly on both versions: CoppeliaSim 4.5.1 and 4.6.0. However, if I try running the example two times in a row, it gets stuck when trying to connect. I tried adding `vi.disconnect_all()`, but it did not solve the problem.

I also downloaded CoppeliaSim 4.2.0-rev5. The best practice example worked without any errors. However, when I open a scene from newer versions of CoppeliaSim, they do not work. To be more specific, I tried opening a scene with a Panda Arm, and the robot became unstable by just pressing Play. I was not expecting it to work properly, anyway. Also, when I open CoppeliaSim 4.2.0 on Ubuntu 22.04, there are some missing functionalities such as robots and scene objects (I mean the ones that were already present in CoppeliaSim 4.2.0, not the newer ones). Maybe it is some incompatibility with Ubuntu 22.04.

I have working code in C++ with the DQ_VrepInterface for my personal work. I could continue using the C++ version. However, we are writing some general purposes code in Python in our lab and it would be good to have some examples in Python for simulation purposes.

As a last test, I will try running the best practice example from @juanjqo on another computer to check if there is any problem with my computer.

I am still thinking about the best option. Also, I need to check with the people from the lab what they think. I do not know yet what will be the best solution for us.

Thanks for the discussion and advice.

marcos-pereira, 2023-11-01:
Hi @juanjqo @mmmarinho ,

I did some more testing here. Somewhy, the best practice example from @juanjqo started working properly on CoppeliaSim 4.5.1-rev4 on my computer. I mean, I can run it multiple times without errors. I tested on another computer here and it also worked. 

A work colleague from the lab suggested the following simple modification to the code to make it run N times. 

```python
#!/bin/python3
from dqrobotics.interfaces.vrep import DQ_VrepInterface
import time

vi = DQ_VrepInterface()

def main() -> None:
    vi = DQ_VrepInterface()
    try:
        if not vi.connect("127.0.0.1", 19997, 100, 10):
            raise RuntimeError("Unable to connect to CoppeliaSim.")

        vi.start_simulation()
        time.sleep(0.1)
        # --------------Your code here----------------------------

        # ---------------------------------------------------------

    except (Exception, KeyboardInterrupt) as e:
        print(e)
        pass

    finally:
        vi.stop_simulation()
        vi.disconnect()

if __name__ == "__main__":
    N = 100
    for i in range(N):            
        main()
        print(f"testing connection: run {i}")
        
```

It worked as expected for N=100. I am not saying this proves that it always works, but using 4.5.1 may be a partial solution for the moment. I will discuss this further inside our group. 

I have no solution in mind for the moment. You can decide on closing the issue or leaving it open.

Thanks!

mmmarinho, 2023-11-01:
@marcos-pereira @juanjqo 

I’ll close this issue because the original purpose was already covered in mine and @juanjqo’s reply.

@marcos-pereira  Feel free to open a new issue as long as the version of CoppeliaSim and `cpp-interface-vrep` precisely match.

closed 2023-11-01 by mmmarinho

marcos-pereira, 2023-11-03:
To anyone who encounters any problem when running CoppeliaSim 4.2.0-rev4, make sure the CoppeliaSim **Edu** was downloaded. The issue I was having with missing robots and interface buttons was because I had wrongly downloaded the CoppeliaSim **Player** (never do anything in a hurry). Also, you may need to add `allowOldEduRelease=7775` to the `system/usrset.txt` inside the CoppeliaSim folder as mentioned [here](https://forum.coppeliarobotics.com/viewtopic.php?t=9334). The DQ_VrepInterface worked correctly as expected on CoppeliaSim 4.2.0-rev4.