dqrobotics/python #54: [LANGUAGE INCOMPATIBILITY] raw_fkm method unavailable for DQ_SerialWholeBody class

ChrisTYuen, 2023-12-05:
**************************
### Code of Conduct
_By submitting this report you automatically agree that you've read and accepted the following conditions._ 
- Support for DQ Robotics is given voluntarily and it's not the developers' role to educate and/or convince anyone of their vision.
- Any possible response and its timeliness will be based on the relevance, accuracy, and politeness of a request and the following discussion.
- If a DQ Robotics member replies, the user must let them know if their response solves their issue or not. 
- Any suggestion/advice/request made by anyone, as well intentioned as they might be, might not be incorporated into DQ Robotics.
*************************

**Describe the missing/unexpected functionality**
When using the DQ_SerialWholeBody robot modeling class, the raw_fkm() method is not supported via python version, but is supported via C++ and Matlab versions. 

**Python behavior** (if applicable)

*PYTHON SCRIPT*

```python
x_jxx = manipulator.raw_fkm(theta, jointx_min)
```
*PYTHON OUTPUT*

```bash
'dqrobotics._dqrobotics._robot_modeling.DQ_SerialWh' object has no attribute 'raw_fkm'.
```
**Environment:**
 - OS: Ubuntu 20.04
 - dqrobotics version 23.4.0a16
 - Python version 3.8.10

**Additional context**
Add any other context about the problem here.

mmmarinho, 2023-12-05:
@ChrisTYuen Thanks!

I understand what you are saying, but for the sake of completeness, could you add a complete [minimal working example](https://en.m.wikipedia.org/wiki/Minimal_reproducible_example)?

ChrisTYuen, 2023-12-06:
@mmmarinho Of course!

I have updated the comment as requested. Let me know if there is anything else.

mmmarinho, 2023-12-06:
@ChrisTYuen Thanks!

@juanjqo The `theta` does not seem correct in the minimal example. Does it also not work with the correct `theta`?

juanjqo, 2023-12-06:
@mmmarinho I was not able to run the minimal example since the module `vs_robot` was not included. 

I did a minimal example to illustrate the problem reported by @ChrisTYuen.

### Minimal Python example:

```python
#!/bin/python3
from dqrobotics.robot_modeling import DQ_SerialWholeBody
from dqrobotics.robots import FrankaEmikaPandaRobot
from dqrobotics.robots import KukaLw4Robot
import numpy as np

def main() -> None:
    try:
        robot = DQ_SerialWholeBody(FrankaEmikaPandaRobot.kinematics())
        robot.add(KukaLw4Robot.kinematics())
        q = np.zeros(robot.get_dim_configuration_space())
        x_raw = robot.raw_fkm(q)
    except (Exception, KeyboardInterrupt) as e:
        print(e)
        pass

if __name__ == "__main__":
    main()
    
```

### Output: 

```shell

'dqrobotics._dqrobotics._robot_modeling.DQ_SerialWh' object has no attribute 'raw_fkm'

Process finished with exit code 0
```

### Expected behavior:

```
1i + E*( - 0.06465 - 0.087j)

Process finished with exit code 0
```

### Minimal example (Matlab):

```matlab
clear all
close all
clc

robot = DQ_SerialWholeBody(FrankaEmikaPandaRobot.kinematics());
robot.add(KukaLwr4Robot.kinematics());
q = zeros(robot.get_dim_configuration_space());
x_raw = robot.raw_fkm(q)
```

### Output (Matlab):
```shell
x_raw = 

         (1i) + E*( - 0.06465 - 0.087j)
```

Best regards, 

Juancho

commit 35f306cf (2023-12-08): [DQ_SerialWholeBody_py.cpp] Added two raw_fkm methods to fix #54 (#55, @juanjqo)

* [DQ_SerialWholeBody_py.cpp] Added two raw_fkm methods to fix #54

* [DQ_SerialWholeBody_py.cpp] Updated the copyright year.