dqrobotics/python #58: [BUG] Unknown issue with usage of the library in general

qlin960618, 2024-07-10:
**************************
### Code of Conduct
_By submitting this report you automatically agree that you've read and accepted the following conditions._ 
- Support for DQ Robotics is given voluntarily and it's not the developers' role to educate and/or convince anyone of their vision.
- Any possible response and its timeliness will be based on the relevance, accuracy, and politeness of a request and the following discussion.
- If a DQ Robotics member replies, the user must let them know if their response solves their issue or not. 
- Any suggestion/advice/request made by anyone, as well intentioned as they might be, might not be incorporated into DQ Robotics.
*************************

**Bug description**
- Unknown issue with library in general. When calling some function result in `SIGSEGV`

**To Reproduce**
- Installing dqrobotics from pip in venv
- Running python console

*Code*
Assume a clean venv is setup, (currently tested with 3.10 on Windows, and 3.9.6 on Mac Apple Silicon)
```shell
python -m pip install dqrobotics
python
```
the pip command default to dqrobotics==23.4.0a22
but I was able to also test 21.4.0a9 with same issue... 

```python
from dqrobotics import *
x = DQ([1])
```
result in segfault

**Environment:**
 - OS: [Windows 11, MacOS Sonoma]
 - dqrobotics version 23.4.0a22,  21.4.0a9, 21.4.0a13
 - Python version 3.10, 3.9.6

juanjqo, 2024-07-10:
@qlin960618 thanks!
@mmmarinho I was able to reproduce the issue. However, when I use the developer workflow (i.e., installing locally with `python3 setup.py build` + `python3 setup.py install`) the code works as expected. Not sure what is going on.

mmmarinho, 2024-07-10:
After re-running the actions, it seems that is was only working on Python 3.8. Why? I don't really know.

After updating pybind to `v2.13.1`, it started working on all versions I could see. We have to wait for a while to see if that propagates to all versions.

Two new problems appeared:

- `macOS-latest` is problematic again (I fix this, then they change it, then I fix it, then they change it...) with their tagging and versioning issues.
- The automatic versioning based on the git tag and commit count is no longer working. The latest versions are being tagged as version `0.0.0`. Please, @juanjqo, could you see if you can figure out why the versioning is not working anymore? 

Some pointers

https://github.com/dqrobotics/python/blob/22a7fcd2cc5199ba4bad8ab397668ef2bfe6959b/setup.py#L85

https://github.com/dqrobotics/python/blob/22a7fcd2cc5199ba4bad8ab397668ef2bfe6959b/setup.py#L13

qlin960618, 2024-07-10:
@mmmarinho Thanks! Please let me know if there is anything I can do to help on my side. 
Additionally, I assume I should keep this issue open for the time being?

mmmarinho, 2024-07-10:
@qlin960618 Yes, it will be open until we get the version thing sorted. It is not the main point but it's clearly a branched problem from that one.

If you could take a look at the versioning thing you're welcome to suggest a solution.

qlin960618, 2024-07-11:
@mmmarinho, about the versioning thing...
Just as debugging step I tried with the command
```shell
python3 setup.py --version
```
not sure if this is related, but this was the output
```shell
/Users/quentinlin/Github/dqrobotics_python/venv/lib/python3.9/site-packages/setuptools/__init__.py:80: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
!!

        ********************************************************************************
        Requirements should be satisfied by a PEP 517 installer.
        If you are using pip, you can try `pip install --use-pep517`.
        ********************************************************************************

!!
  dist.fetch_build_eggs(dist.setup_requires)
/Users/quentinlin/Github/dqrobotics_python/venv/lib/python3.9/site-packages/setuptools/_distutils/dist.py:268: UserWarning: Unknown distribution option: 'version_config'
  warnings.warn(msg)
0.0.0
```
specifically the part about `Unknown distribution option: 'version_config'` warning?

qlin960618, 2024-07-11:
@mmmarinho, a proposed solution i think... but it is a bit of a hacks,
its what I have been using for jenkins
```python
try:
    import subprocess
    tag_cmd = "git describe --tags --abbrev=0"
    ret = subprocess.run(tag_cmd, shell=True, check=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    tags = ret.stdout.decode("utf-8").strip().split(".")
    try:
        version_major = int(tags[0])
        version_minor = int(tags[1])
        version_patch = int(tags[2])
    except ValueError as e:
        raise ValueError("Invalid tag format")
    print("setup.py LOG: version: ", version_major, version_minor, version_patch)
    try:
        get_branch_cmd = "git symbolic-ref --short HEAD"
        ret = subprocess.run(get_branch_cmd, shell=True, check=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
        branch = ret.stdout.decode("utf-8").strip()
    except subprocess.CalledProcessError as e:
        print("Cannot get branch name from cmd")
        branch = os.environ["GIT_BRANCH"].  #### JENKINS gives this as env variable
    print("setup.py LOG: branch: ", branch)
    # print(tags, branch)
    build_num_cmd = "git rev-list HEAD --count"
    ret = subprocess.run(build_num_cmd, shell=True, check=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    build_number = int(ret.stdout.decode("utf-8").strip())
    print("setup.py LOG: build_number: ", build_number)
    if branch.find("master") >= 0:  # alpha
        __version__ = f"{version_major}.{version_minor}.{version_patch}a{build_number}"
    elif branch.find("release") >= 0:
        __version__ = f"{version_major}.{version_minor}.{version_patch}.{build_number}"
    else:
        raise ValueError("Unknown branch name")
except ValueError as e:
    raise ValueError("Cannot get version number from git tags")
```

mmmarinho, 2024-07-11:
@juanjqo , @qlin960618 

Sorry for the cryptic request.

The dqrobotics python library is using https://pypi.org/project/setuptools-git-versioning/1.7.3/ to figure out the necessary version information from git.

Could you please check if
- They changed how that works? If you install locally and the version is not weird, then it's not it possibly.
- The GitHub Actions is not checking out the repository with the necessary history. This could be the case.

qlin960618, 2024-07-11:
@mmmarinho 
ha... I think maybe they just change the argument name
I updated the `version_config` to `setuptools_git_versioning` it seem to work locally. 

https://github.com/dqrobotics/python/blob/22a7fcd2cc5199ba4bad8ab397668ef2bfe6959b/setup.py#L85

```shell
(venv) ➜  dqrobotics_python git:(master) ✗ python setup.py --version
/Users/quentinlin/Github/dqrobotics_python/venv/lib/python3.9/site-packages/setuptools/__init__.py:80: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
!!

        ********************************************************************************
        Requirements should be satisfied by a PEP 517 installer.
        If you are using pip, you can try `pip install --use-pep517`.
        ********************************************************************************

!!
  dist.fetch_build_eggs(dist.setup_requires)
23.4.0a25
```

mmmarinho, 2024-07-12:
@qlin960618 The second suggestion seems to be the case. I'll try it out and see where that takes us.

Eventually we need to jump into `setuptools`, as described in version `2.0.0`, https://pypi.org/project/setuptools-git-versioning/2.0.0/, but let's do that when the needs justify the effort.

mmmarinho, 2024-07-12:
Solved.

closed 2024-07-12 by mmmarinho

mmmarinho, 2024-07-12:
>     * `macOS-latest` is problematic again (I fix this, then they change it, then I fix it, then they change it...) with their tagging and versioning issues. PS: `macOS-latest` now is using an M1 machine, which explains the problem. See screenshot attached because they will eventually update the website.
> 

Fixed the `macos-latest` issue in 363627c. They are now built in Github-hosted machines, and the self-hosted ones are no longer needed.