dqrobotics/python #59: [LANGUAGE INCOMPATIBILITY] Missing default value for damping in DQ_ClassicQPController for Python

anachristinaac, 2024-07-30:
**Description of the missing/unexpected functionality**

When using the DQ_ClassicQPController, the MATLAB version defines a default value for the damping variable, but the Python version does not.

**Python behavior**

*PYTHON SCRIPT*

```python
from dqrobotics import *
from dqrobotics.solvers import DQ_QuadprogSolver
from dqrobotics.robot_control import DQ_ClassicQPController
from dqrobotics.robots import KukaLw4Robot

robot = KukaLw4Robot.kinematics()
solver = DQ_QuadprogSolver()
controller = DQ_ClassicQPController(robot, solver)
controller.get_damping()
```

*PYTHON OUTPUT*

```bash
7.73974598944389e+228
```

**Matlab behavior**

*MATLAB SCRIPT*

```matlab
robot = KukaLwr4Robot.kinematics();
solver = DQ_QuadprogSolver;
controller = DQ_ClassicQPController(robot, solver);
controller.get_damping()
```

*MATLAB OUTPUT*

```bash
1.0000e-03
```

**Environment:**
 - OS: Ubuntu 22.04.4 LTS
 - dqrobotics version MATLAB: commit [e455c66](https://github.com/dqrobotics/matlab/commit/e455c663a51a065cce7cc7d8884e6c2a2eb1332b)
 - MATLAB version: R2023b
 -  dqrobotics version Python: 23.4.0a22
 - Python version: 3.10.12

juanjqo, 2024-07-31:
@anachristinaac thank you for let us know. It looks like it is an issue in the C++ version. 

@mmmarinho I proposed a [PR in C++](https://github.com/dqrobotics/cpp/pull/67) to fix this language incompatibility. I tested both C++ and Python locally. 

### Minimal example:

```python
from dqrobotics.solvers import DQ_QuadprogSolver
from dqrobotics.robot_control import DQ_ClassicQPController
from dqrobotics.robots import KukaLw4Robot

def main() -> None:
    robot = KukaLw4Robot.kinematics()
    solver = DQ_QuadprogSolver()
    controller = DQ_ClassicQPController(robot, solver)
    print(controller.get_damping())

if __name__ == "__main__":
    main()
```

### Output:

```shell
0.001
```

```cpp
#include <iostream>
#include <dqrobotics/robots/KukaLw4Robot.h>
#include <dqrobotics/robot_control/DQ_ClassicQPController.h>
#include <dqrobotics/solvers/DQ_PROXQPSolver.h>

using namespace DQ_robotics;

int main()
{
    auto robot = std::make_shared<DQ_SerialManipulatorDH>(KukaLw4Robot::kinematics());
    auto solver = std::make_shared<DQ_PROXQPSolver>();
    auto controller = DQ_ClassicQPController(robot, solver);
    auto damping = controller.get_damping();
    std::cout<<damping<<std::endl;
}
```

### Output:

```shell
0.001
```

mmmarinho, 2024-09-26:
Fixed with https://github.com/dqrobotics/cpp/pull/67.

If needed, please re-open the issue.

closed 2024-09-26 by mmmarinho