dqrobotics/python #61: [BUG] The connect method using default arguments is not available

juanjqo, 2025-03-03:
**************************
### Code of Conduct
_By submitting this report you automatically agree that you've read and accepted the following conditions._ 
- Support for DQ Robotics is given voluntarily and it's not the developers' role to educate and/or convince anyone of their vision.
- Any possible response and its timeliness will be based on the relevance, accuracy, and politeness of a request and the following discussion.
- If a DQ Robotics member replies, the user must let them know if their response solves their issue or not. 
- Any suggestion/advice/request made by anyone, as well intentioned as they might be, might not be incorporated into DQ Robotics.
*************************

**Bug description**

@dqrobotics/developers 

Hi @mmmarinho, 

The connect method of the `DQ_CoppeliaSimInterfaceZMQ` class is not available with default arguments. This is happening because some bindings are missing. Specifically, the following C++ method is not exposed in Python:

```cpp
bool connect(const std::string& host = "localhost", const int& port = 23000, const int&TIMEOUT_IN_MILISECONDS = 300) override;
```

*Minimal Example*

```python
from dqrobotics.interfaces.coppeliasim import DQ_CoppeliaSimInterfaceZMQ

def main() -> None:
    vi = DQ_CoppeliaSimInterfaceZMQ()
    try:
        vi.connect() 
        vi.start_simulation()
        vi.stop_simulation()

    except (Exception, KeyboardInterrupt) as e:
        print(e)
        pass

if __name__ == "__main__":
    main()
```
*Output*
```shell
connect(): incompatible function arguments. The following argument types are supported:
    1. (self: dqrobotics._dqrobotics._interfaces._coppeliasim.DQ_CoppeliaSimInterfaceZMQ, arg0: str, arg1: int, arg2: int, arg3: int) -> bool
```

**Expected behavior**
- A successful connection with CoppeliaSim in "localhost" on the default port 23000.

*Expected output*
```
Process finished with exit code 0
```

**Environment:**
 - OS: Ubuntu 24.04 LTS
 - dqrobotics version 23.4.0a45
 - Python version: 3.12

bvadorno, 2025-03-03:
Thanks, @juanjqo.

If you know how to fix the bug, I suggest you create a pull request and ask Murilo to review it. It'll be faster this way. The same reasoning is valid for the other pull requests.

Kind regards,
Bruno

juanjqo, 2025-03-03:
Hi @bvadorno,

Yes, I'm preparing the PR to fix the bugs. 

Kind regards,

Juancho

commit c2acba1d (2025-03-04): Add `set_stepping_mode`, `get_object_handles`, remove extra string from signatures, and remove protected C++ methods.  Fix #62, Fix #61, and Fix #60 (#63)

* [DQ_CoppeliaSimInterfaceZMQ_py.cpp] Updated the copyright year and added the default connect method.

* [DQ_CoppeliaSimInterfaceZMQ_py.cpp] Added the set_stepping_mode.

* [DQ_CoppeliaSimInterfaceZMQ_py.cpp] Added the methdos get_object_handles{s}.

* [DQ_CoppeliaSimInterfaceZMQ_py.cpp] Removed the bindings of protected C++ methods as set_joint_position, get_joint_position.

* [DQ_CoppeliaSimInterfaceZMQ_py.cpp] Fixed the signature of som [... cut]
closed 2025-03-04 by mmmarinho