dqrobotics/python #64: Actions runner issues with `boost` on Windows

mmmarinho, 2025-04-29:
Hi @juanjqo,

The actions runner is having a hard time installing `boost` for Windows. Unfortunately, `cpp-zmq` needs it, so we need to have it installed.

https://github.com/dqrobotics/python/blob/d0e610c31fca37b47db12f0ebcf942dc1861c440/.github/workflows/python_package.yml#L57

In a Windows environment, could you please check which packages of Boost are needed? Installing the entirety of `boost` with `vcpkg` is taking more than the time allocated for our free runners.

Kind regards,
Murilo

juanjqo, 2025-04-29:
@mmmarinho sure!

mmmarinho, 2025-06-17:
Hi @juanjqo, any information on this topic? 

If the necessary packages in `.\vcpkg` are identified this installation becomes streamlined in Windows. Everything else is working fine.